familab/LabTasks #10: Bathrooms

bradsilcox, 2015-07-22:
Bathroom Decor & Accessories
Soap dispensers (liquid soap or liquid soap foamers)
Toilet roll holders (non-mounted; 5 roll storage)
Wall mounted mirrors (I need this)
Trash cans

DaveCasey, 2015-07-22:
Do we have a style in mind?
If I showed up with a couple of something like this:
http://www.amazon.com/Kimberly-Clark-Professional-ScottFold-Stainless-Dispenser/dp/B0040ZOE8A
or this
http://www.amazon.com/TrippNT-51912-Dispensing-Multi-Fold-Protective/dp/B004S7SPXK
and maybe this
http://www.amazon.com/Simply-Silver-Wall-Mount-Soap-Dispenser/dp/B00C19SDL6

on Saturday, would that be helpful?  Who should I work with regarding wall penetration?

hybridsix, 2015-07-22:
Paper towel holders are already installed. :) We need some soap dispensers though, and toilet paper holders and such. We have some things that can be returned to home depot and ikea for store credit if you plan on visiting some of those stores.

willasaywhat, 2015-07-22:
If we need anything in this realm I'm happy to pitch in or pick stuff up. Sounds like Dave has a plan though. I'll just wait and make it pretty afterwards. :p

closed 2016-02-01 by willasaywhat