familab/LabTasks #5: Restroom Signs

bradsilcox, 2015-07-22:
I love the walk sign we have, but I want to make it smaller. I was thinking something thin, that normally would have white led's reading Restroom or some such. It would look like a sign you would see in a restaurant but lit. Then when the light was on in the restroom it switch to in use (red/orange). I believe the wiring is super simple except having to convert 2 sources of AC (line and switch) to dc.

willasaywhat, 2015-09-08:
A temporary workaround has been added (labels). This can be closed for now.

npolanosky, 2016-02-01:
Do we still just have labels? If so, it would be super easy to make nicer lit signs and I could do it tomorrow.

lancehudson, 2016-02-01:
yes just labels, ones that indicate occupancy?

npolanosky, 2016-02-01:
That would be super cool. Will figure out how to make it do.