familab/LabTasks #14: Access Control System

willasaywhat, 2015-07-23:
* Develop hardware solution for authn
* Develop software backend for authn/authz
* Develop membership management system frontend
* Payment integration
** Stripe (See: https://support.stripe.com/questions/can-i-pull-my-stripe-data-quickbooks and https://stripe.com/subscriptions)

willasaywhat, 2015-07-23:
Lance has the first release of the door hardware and code ready for testing. 

The membership management side of things should keep track of memberships and activate/deactivate key cards as well as provide authorization for restricted tools like the laser, biohacking room, and the Haas. 

I checked briefly into Stripe, as they have a good subscription management system with a solid API. Ideally we would have a system that integrated with PayPal and had a method for people to pay by check automatically. Big thing is tracking and Quickbooks integration.

hybridsix, 2015-07-27:
I will be bringing material and a file to laser-cut the first covers and mounts for the door lock system. I'll upload the files here once they're more complete.