familab/LabTasks #17: Audio System

hybridsix, 2015-07-27:
System needs termination in Network/AV Closet. Some cables already there, some will need to be made, and some will need to be toned out. 

The room cables are not exposed at the moment, and need to be with a low-voltage cut in plate. 

Priority is to get speakers working again with a Bluetooth receiver. Maybe one for inside, one for outside to start with. 

Program needs to be written for X-Frame, and incorporate future wall controls and zones. Will include the 8802 for additional inputs.

lancehudson, 2016-02-01:
lots of this is done, @hybridsix  is still working on final programming and additional inputs.

npolanosky, 2016-12-07:
I believe this is complete or has stalled and is complete enough.

closed 2016-12-07 by npolanosky