familab/LabTasks #39: Famduino is down; last image 3/31 @0231

darkmoonsinger, 2016-04-07:


willasaywhat, 2016-04-07:
If someone is there before I get back please attempt to power cycle the affected devices (cameras, pi). Failing that, check that the DHCP reservations are in place for Black, White, and famduino pi. Failing all of that, I can look at it when I get back. This has been happening a lot; not sure what has changed.

waterbury, 2016-04-12:
Pi, and cameras are connected. Scripts run, and images are taken, and stored. Script appears to fail uploading to imgur. 

From /tmp/white.log:

/usr/local/lib/python2.7/dist-packages/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
Camera panned to leftmost stop.
Snapshot 1 done.
Snapshot 2 done.
Snapshot 3 done.
Snapshot 4 done.
Snapshot 5 done.
Snapshot 6 done.
Snapshot 7 done.
Traceback (most recent call last):
  File "/home/pi/famduino/white/py_foscam_labcam/pan_and_tweet.py", line 38, in <module>
    url = upload_to_imgur(file, title, sec["imgur_id"], sec["imgur_apikey"])
  File "/home/pi/famduino/white/py_foscam_labcam/functions.py", line 58, in upload_to_imgur
    data_back = json.loads(j1.content)
  File "/usr/lib/python2.7/json/__init__.py", line 326, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 365, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 383, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded

waterbury, 2016-04-15:
Imgur API  key confirmed to work on external IP

waterbury, 2016-04-15:
Email from Imgur on Fri, 04/15/2016:

Thank you for writing in. I apologize for the delay. I have been working with the team to find the cause of this as we do not block any IP addresses. 

We have a fix that is going out, unfortunately it does take time to implement. The fix should be accomplished by late monday, early tuesday.

Again we apologize for this inconvenience. 
We are always here to help if you have any other questions, comments or concerns.

Thank you,

Isaac K.
Take the joy of Imgur with you wherever you go with the Imgur apps! http://imgur.com/apps

willasaywhat, 2016-05-11:
This has been resolved permanently by switching to S3.

closed 2016-05-11 by willasaywhat